capntrips/KernelFlasher #1: Stuck at the "Google" screen after flashing

powellnorma, 2023-01-14:
Using FKM instead worked fine. Others seem to have the same problem:

https://forum.xda-developers.com/t/kernel-blu_spark-r171-a13-jan-2023.4182455/post-87337427
https://forum.xda-developers.com/t/kernel-blu_spark-r171-a13-jan-2023.4182455/post-87940883

capntrips, 2023-01-14:
I looked at the log posted in one of your links and didn't see anything obvious.

I don't have a Pixel 4a, so it will be hard for me to test. Does it happen every time, or is it intermittent?

From a glance at the kernel zip, boot is the only partition modified. Would you be willing to post a dump of the stock partition and the modified one after flashing both with Kernel Flasher and FKM? I'm not sure why the latter two would be different, since the actual write should be handled by AK3 in both cases, but it might be a good first step.

powellnorma, 2023-01-16:
> the KernelFlasher app crashes and can't be opened again

Once I flash the (working) backup of stock-rooted manually via dd, KernelFlasher opens again just fine

capntrips, 2023-01-16:
Thanks for the files. I've got them downloaded and will take a peek at them shortly.

> the KernelFlasher app crashes and can't be opened again

It sounds like it's flashing a bad image to boot and can't read that image on startup. Does it open to a error screen, or is it a hard crash?

Once the issue with flashing goes away, the inability to open the app should as well, but if you'll provide a logcat during the crash, I can try to catch the specific issue and at least display a useful error message.

If you reflash in KernelFlasher to get the crash logcat, the flash log from FKM when you revert might also be useful.

capntrips, 2023-01-16:
I'm able to unpack the KernelFlasher flashed image, but magiskboot throws `bad cpio header` when attempting to test or extract the ramdisk. That same error is sprinkled through your AK3 log as well.

I'm curious if you can can copy, unpack, and extract the files if performed directly from the terminal. You can test it with the following commands:

```
cd /data/user/0/com.github.capntrips.kernelflasher/files
rm -rf akhome
mkdir akhome
cd akhome
dd if=/dev/block/by-name/boot_b of=boot.img
/data/adb/magisk/magiskboot unpack boot.img
/data/adb/magisk/magiskboot cpio ramdisk.cpio sha1
```
If for some reason it doesn't work there, can you also try in `/data/local/tmp` as well?

powellnorma, 2023-01-17:
Thank you for investigating

> Does it open to a error screen, or is it a hard crash?

A hard crash, I will attach the logcat in the .zip

> the flash log from FKM when you revert might also be useful

I have attached the FKM log from 'stock-root -> custom'

> I'm curious if you can can copy, unpack, and extract the files if performed directly from the terminal.

stock-rooted -> yes
KernelFlasher -> no (bad cpio header)
FKM -> yes

> what is the output of `mount | grep -E ' /data |/data/user|/data/local'`?

```
/dev/block/dm-36 on /data type f2fs (rw,lazytime,seclabel,nosuid,nodev,noatime,background_gc=on,discard,no_heap,user_xattr,inline_xattr,acl,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,reserve_root=32768,resuid=0,resgid=1065,alloc_mode=default,checkpoint_merge,fsync_mode=nobarrier,discard_unit=block)
/dev/block/dm-36 on /data/user/0 type f2fs (rw,lazytime,seclabel,nosuid,nodev,noatime,background_gc=on,discard,no_heap,user_xattr,inline_xattr,acl,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,reserve_root=32768,resuid=0,resgid=1065,alloc_mode=default,checkpoint_merge,fsync_mode=nobarrier,discard_unit=block)
```

Link: https://wormhole.app/ZnXNL#956g75dPB_QWUvvMszEgiQ

capntrips, 2023-01-19:
I didn't get to your zip in time, if you wouldn't mind sharing it again. I'll be out of pocket for two days beginning in the morning, so if it doesn't let you set an expiration beyond that, we may need to wait until I return or use a different file host.

Maybe we can try running the AK3 zip directly, just to rule out something with the environment (assuming the zip is in `/sdcard/Download`):

```
cd /data/user/0/com.github.capntrips.kernelflasher/files
rm -rf akhome
mkdir akhome
unzip -p /sdcard/Download/blu_spark_r171-pxl4a_0d9a2328c.zip META-INF/com/google/android/update-binary > akhome/update-binary
AKHOME=$(pwd)/akhome $SHELL ./akhome/update-binary 3 1 /sdcard/Download/blu_spark_r171-pxl4a_0d9a2328c.zip
```

You may have also included it in the previous zip, but might be interesting to see the logcat during a failed flash, as well.

powellnorma, 2023-01-23:
> just to rule out something with the environment

When I execute those commands I get:

```
./akhome/update-binary: line 31: /proc/self/fd/: Is a directory
./akhome/update-binary: line 22: /proc/self/fd/: Is a directory
unzip: couldn't open : I/O error
ls: tools: No such file or directory
./akhome/update-binary: line 22: /proc/self/fd/: Is a directory
```

New Link: https://send.q1q.wtf/download/db7608ac460bc63c/#ANDNn9RQbjYxdxJgmB4U3w

capntrips, 2023-01-31:
Sorry for the delay in getting back to you on this. I just returned home from nearly two weeks on the road yesterday and am still playing a bit of catch up.

Now that I'm home, I have access to a Pixel 6a to run a few tests. I probably forgot a step somewhere with running AK3 directly. I'll have to track down the script I was using before I wrote Kernel Flasher.

capntrips, 2023-02-07:
I couldn't immediately find my old version, but found the [script](https://forum.xda-developers.com/t/tools-zips-scripts-osm0sis-odds-and-ends-multiple-devices-platforms.2239421/post-79928640) that was the basis for that script that developed into Kernel Flasher. I reworked it to basically the same thing Kernel Flasher does in the background.

[flash_ak3.sh.gz](https://github.com/capntrips/KernelFlasher/files/10679394/flash_ak3.sh.gz)

I [reduced](https://github.com/capntrips/KernelFlasher/files/10679342/generic_dump.patch.gz) the [latest](https://github.com/osm0sis/AnyKernel3/tree/950b8e57118288f4df9341c45cbb91e3939782e7) stock AK3 to just the `dump_boot` command, which will simply split out and unpack the ramdisk. We'll "flash" this file to verify AK3 works in the same context outside of the app.

[UPDATE-AnyKernel3.zip](https://github.com/capntrips/KernelFlasher/files/10679313/UPDATE-AnyKernel3.zip)

Once you have those two files, run the following commands:

```
adb push flash_ak3.sh.gz /data/local/tmp
adb push UPDATE-AnyKernel3.zip /sdcard/Download
adb shell
su
cd /data/local/tmp
gunzip flash_ak3.sh.gz
chmod +x flash_ak3.sh
./flash_ak3.sh
```
In theory, you should get output similar to [this](https://github.com/capntrips/KernelFlasher/files/10679454/ak3-log--2023-02-07--15-15.log).

capntrips, 2023-02-15:
I realized `dump_boot` may not pass through any errors, so I [added](https://github.com/capntrips/KernelFlasher/files/10738916/dump_test_sha1.patch.gz) a test and sha1 to the operation. I also dropped the attributes function to silence the output a bit.

[UPDATE-AnyKernel3.zip](https://github.com/capntrips/KernelFlasher/files/10738923/UPDATE-AnyKernel3.zip)

powellnorma, 2023-02-18:
Hey, this seems to work fine, here are the logs I got: [logs.zip](https://github.com/capntrips/KernelFlasher/files/10774764/logs.zip)

capntrips, 2023-02-19:
There's obviously no issue with unpacking the image or extracting the ramdisk in Kernel Flasher on your device. It's weird that it has trouble in the write phase. It might be interesting to patch `blu_spark_r171-pxl4a_0d9a2328c.zip` with the relevant lines from that script, just to confirm it is able to read everything before it tries to flash the image.

[blu_spark_r171-1-pxl4a.patch.gz](https://github.com/capntrips/KernelFlasher/files/10776317/blu_spark_r171-1-pxl4a.patch.gz)

Assuming you have the above patch and `blu_spark_r171-pxl4a_0d9a2328c.zip` in your working directory, you should be able to run the following commands:

```
gunzip -k blu_spark_r171-1-pxl4a.patch.gz
mkdir blu_spark_r171-pxl4a_0d9a2328c
cd blu_spark_r171-pxl4a_0d9a2328c
unzip ../blu_spark_r171-pxl4a_0d9a2328c.zip
patch -p1 < ../blu_spark_r171-1-pxl4a.patch
zip -r9 ../blu_spark_r171-1-pxl4a-debugging.zip * -x .git README.md *placeholder
cd ..
rm -r blu_spark_r171-pxl4a_0d9a2328c
adb push blu_spark_r171-1-pxl4a-debugging.zip /sdcard/Download/
```

**Note:** This one will actually flash blu_spark r171 to your device.

I patched `blu_spark_r171-pxl4a_0d9a2328c.zip` with the below patch and dropped the stock rooted `boot.img` from the backups you shared with me previously (renamed to `block.img` to avoid a conflict with AK3) into the zip. I flashed it with Kernel Flasher, and it completed without issue, and the resulting `target.img` matched almost exactly to the backup you shared with me after flashing with FKM. I would be curious to see if that one completes without error on your device, as well.

[blu_spark_r171-2-pxl4a.patch.gz](https://github.com/capntrips/KernelFlasher/files/10776319/blu_spark_r171-2-pxl4a.patch.gz)

Assuming you have the above patch, `blu_spark_r171-pxl4a_0d9a2328c.zip`, and your `backups` folder all in your working directory, you should be able to run the following commands:

```
gunzip -k blu_spark_r171-2-pxl4a.patch.gz
mkdir blu_spark_r171-pxl4a_0d9a2328c
cd blu_spark_r171-pxl4a_0d9a2328c
unzip ../blu_spark_r171-pxl4a_0d9a2328c.zip
patch -p1 < ../blu_spark_r171-2-pxl4a.patch
cp ../backups/stock-rooted/boot.img block.img
zip -r9 ../blu_spark_r171-2-pxl4a-debugging.zip * -x .git README.md *placeholder
cd ..
rm -r blu_spark_r171-pxl4a_0d9a2328c
adb push blu_spark_r171-2-pxl4a-debugging.zip /sdcard/Download/
```

**Note:** This one will not actually flash anything, as it will create a `/data/local/tmp/target.img` instead.

If you have any trouble generating the zips, let me know, and maybe I can share them with you directly via DM on XDA or elsewhere. I would rather not host blu_spark test builds publicly, if only to avoid the possibility of someone ending up with them by accident.

Since there were no major changes to the relevant file in blu_spark r172, you can apply those same patches to it instead, if you prefer. I used r171 in my testing just to make it easier to compare against your logs.

**Note:** The `-debugging` suffix on the above zips will cause AK3 to write a timestamped `debug.tgz` to the root of your `/sdcard`. These might be worth sharing, if you run into any errors while flashing the zips.

Sorry for the wall of text, and hopefully I didn't miss anything. Thanks for your help with debugging this.

capntrips, 2023-02-22:
Someone shared a `debug.tgz` on XDA, so there's no need to run the above test. It looks like the ramdisk gets unpacked and extracted without issue, so something must be going wrong later in the process.

capntrips, 2023-02-23:
Since the ramdisk is fine [here](https://github.com/osm0sis/AnyKernel3/blob/398b9631df31228cb2d2613c9b370767a457b200/tools/ak3-core.sh#L174) and [here](https://github.com/osm0sis/AnyKernel3/blob/398b9631df31228cb2d2613c9b370767a457b200/anykernel.sh#L73) but throws an error [here](https://github.com/osm0sis/AnyKernel3/blob/398b9631df31228cb2d2613c9b370767a457b200/tools/ak3-core.sh#L214), I'm guessing the issue is [here](https://github.com/osm0sis/AnyKernel3/blob/398b9631df31228cb2d2613c9b370767a457b200/tools/ak3-core.sh#L209) or [here](https://github.com/osm0sis/AnyKernel3/blob/398b9631df31228cb2d2613c9b370767a457b200/tools/ak3-core.sh#L179). The `ramdisk.cpio` with the `bad cpio header` ends up in the `debug.tgz` because the bad one is copied [here](https://github.com/osm0sis/AnyKernel3/blob/398b9631df31228cb2d2613c9b370767a457b200/tools/ak3-core.sh#L310).

With the kernel zip and a good `boot.img` in your /sdcard/Download folder, can you run the following:

```
adb shell
su
cd /data/local/tmp
mkdir {bin,ramdisk}
cd bin
unzip -j /sdcard/Download/blu_spark_r171-pxl4a_0d9a2328c.zip tools/{busybox,magiskboot}
PATH="$(pwd):$PATH"
busybox --install -s .
cd ..
magiskboot unpack -h /sdcard/Download/boot.img
cd ramdisk
EXTRACT_UNSAFE_SYMLINKS=1 cpio -d -F ../ramdisk.cpio -i
find . | cpio -H newc -o > ../ramdisk-new.cpio
magiskboot cpio ../ramdisk-new.cpio test
```

I intentionally left out the bits copying the ramdisk files from the kernel zip to isolate it, so if the above works without a `bad cpio header` error, run the following in the same `adb` shell:

```
cd ..
unzip /sdcard/Download/blu_spark_r171-pxl4a_0d9a2328c.zip "ramdisk/*"
cd ramdisk
find . | cpio -H newc -o > ../ramdisk-new2.cpio
magiskboot cpio ../ramdisk-new2.cpio test
```

capntrips, 2023-02-23:
The issue has been isolated. See [here](https://www.mail-archive.com/toybox@lists.landley.net/msg07206.html), [here](https://bugs.busybox.net/show_bug.cgi?id=12341), [here](https://github.com/google/fscrypt/issues/305), [here](8828d9cf77ec57604925753eeac45d0463a5b74f), and [here](https://github.com/torvalds/linux/commit/d18760560593e5af921f51a8c9b64b6109d634c2) for more info. AK3 uses busybox rather than toybox, but it was affected in the same way and hasn't been patched. It is inconsistent between devices as it was patched in the kernel, and some devices may not have gotten the patch.

The solution was to basically [chroot](https://forum.xda-developers.com/t/tools-zips-scripts-osm0sis-odds-and-ends-multiple-devices-platforms.2239421/post-79928640) the flashing environment, which works around the issue. FKM and likely EXKM were already doing this or something very similar, which is why they weren't affected.

A build is being tested (see the busybox branch for the changes) and will be posted soon. I'll close this issue when I merge it into main.

capntrips, 2023-02-23:
v1.0.0-alpha12+tempfs with the fix included is on the releases page. I'll close this issue when I merge it into main.

capntrips, 2023-03-21:
fixed in c834f5f and released in [v1.0.0-alpha14](https://github.com/capntrips/KernelFlasher/releases/tag/v1.0.0-alpha14)

closed 2023-03-21 by capntrips