capntrips/KernelFlasher #9: Crash on opening

xFlaem, 2023-03-30:
**Redmi Note 8T android 11**
`java.lang.RuntimeException: Error receiving broadcast Intent { act=c4226974-2de7-43a1-b535-3c1faea5e712 flg=0x400010 pkg=com.github.capntrips.kernelflasher (has extras) } in com.topjohnwu.superuser.internal.RootServiceManager$ServiceReceiver@ec3cab9
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1572)
	at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:236)
	at android.app.ActivityThread.main(ActivityThread.java:8037)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:656)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:967)
Caused by: java.lang.NullPointerException
	at com.github.capntrips.kernelflasher.ui.screens.main.MainViewModel.<init>(MainViewModel.kt:59)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2$1.invoke(MainActivity.kt:170)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2$1.invoke(MainActivity.kt:168)
	at androidx.lifecycle.viewmodel.InitializerViewModelFactory.create(InitializerViewModelFactory.kt:107)
	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:187)
	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:153)
	at androidx.lifecycle.viewmodel.compose.ViewModelKt.get(ViewModel.kt:215)
	at androidx.lifecycle.viewmodel.compose.ViewModelKt.viewModel(ViewModel.kt:156)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2.invoke(MainActivity.kt:317)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2.invoke(MainActivity.kt:166)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.ui.platform.ComposeView.Content(ComposeView.android.kt:404)
	at androidx.compose.ui.platform.AbstractComposeView$ensureCompositionCreated$1.invoke(ComposeView.android.kt:250)
	at androidx.compose.ui.platform.AbstractComposeView$ensureCompositionCreated$1.invoke(ComposeView.android.kt:249)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.ui.platform.CompositionLocalsKt.ProvideCommonCompositionLocals(CompositionLocals.kt:177)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt$ProvideAndroidCompositionLocals$3.invoke(AndroidCompositionLocals.android.kt:123)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt$ProvideAndroidCompositionLocals$3.invoke(AndroidCompositionLocals.android.kt:122)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt.ProvideAndroidCompositionLocals(AndroidCompositionLocals.android.kt:114)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1$3.invoke(Wrapper.android.kt:157)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1$3.invoke(Wrapper.android.kt:156)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1.invoke(Wrapper.android.kt:156)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1.invoke(Wrapper.android.kt:140)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.ActualJvm_jvmKt.invokeComposable(ActualJvm.jvm.kt:78)
	at androidx.compose.runtime.ComposerImpl$doCompose$2$5.invoke(Composer.kt:3248)
	at androidx.compose.runtime.ComposerImpl$doCompose$2$5.invoke(Composer.kt:3238)
	at androidx.compose.runtime.SnapshotStateKt__DerivedStateKt.observeDerivedStateRecalculations(DerivedState.kt:341)
	at androidx.compose.runtime.SnapshotStateKt.observeDerivedStateRecalculations(Unknown Source:1)
	at androidx.compose.runtime.ComposerImpl.doCompose(Composer.kt:3238)
	at androidx.compose.runtime.ComposerImpl.composeContent$runtime_release(Composer.kt:3173)
	at androidx.compose.runtime.CompositionImpl.composeContent(Composition.kt:587)
	at androidx.compose.runtime.Recomposer.composeInitial$runtime_release(Recomposer.kt:950)
	at androidx.compose.runtime.CompositionImpl.setContent(Composition.kt:519)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:140)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:131)
	at androidx.compose.ui.platform.AndroidComposeView.setOnViewTreeOwnersAvailable(AndroidComposeView.android.kt:1060)
	at androidx.compose.ui.platform.WrappedComposition.setContent(Wrapper.android.kt:131)
	at androidx.compose.ui.platform.WrappedComposition.onStateChanged(Wrapper.android.kt:182)
	at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.java:360)
	at androidx.lifecycle.LifecycleRegistry.addObserver(LifecycleRegistry.java:202)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:138)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:131)
	at androidx.compose.ui.platform.AndroidComposeView.onAttachedToWindow(AndroidComposeView.android.kt:1147)
	at android.view.View.dispatchAttachedToWindow(View.java:20598)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3511)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3518)
	at android.view.ViewGroup.addViewInner(ViewGroup.java:5305)
	at android.view.ViewGroup.addView(ViewGroup.java:5091)
	at android.view.ViewGroup.addView(ViewGroup.java:5060)
	at com.android.internal.policy.PhoneWindow.setContentView(PhoneWindow.java:490)
	at android.app.Activity.setContentView(Activity.java:3621)
	at androidx.activity.ComponentActivity.setContentView(ComponentActivity.java:462)
	at androidx.activity.compose.ComponentActivityKt.setContent(ComponentActivity.kt:68)
	at androidx.activity.compose.ComponentActivityKt.setContent$default(ComponentActivity.kt:49)
	at com.github.capntrips.kernelflasher.MainActivity.onAidlConnected(MainActivity.kt:166)
	at com.github.capntrips.kernelflasher.MainActivity$AidlConnection.onServiceConnected(MainActivity.kt:75)
	at com.topjohnwu.superuser.internal.RootServiceManager.lambda$bindInternal$3(RootServiceManager.java:241)
	at com.topjohnwu.superuser.internal.RootServiceManager$$ExternalSyntheticLambda1.run(Unknown Source:6)
	at com.topjohnwu.superuser.internal.UiThreadHandler.run(UiThreadHandler.java:31)
	at com.topjohnwu.superuser.internal.UiThreadHandler$$ExternalSyntheticLambda0.execute(Unknown Source:0)
	at com.topjohnwu.superuser.internal.RootServiceManager.bindInternal(RootServiceManager.java:241)
	at com.topjohnwu.superuser.internal.RootServiceManager.lambda$createBindTask$5$com-topjohnwu-superuser-internal-RootServiceManager(RootServiceManager.java:257)
	at com.topjohnwu.superuser.internal.RootServiceManager$$ExternalSyntheticLambda4.run(Unknown Source:8)
	at com.topjohnwu.superuser.internal.RootServiceManager$ServiceReceiver.onReceive(RootServiceManager.java:400)
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1562)
	... 8 more
`

capntrips, 2023-03-30:
Are you on the latest build?
Is Redmi Note 8T an A/B device?
If both are yes, can you tell me where to find your `boot_a` block device? For example, `/dev/block/by-name/` or `/dev/block/mapper/`?

P.S. If you'll use triple backticks instead of single, it'll preserve the whitespace in your code block and make it a bit easier to read the backtrace. See [here](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) for more details.

capntrips, 2023-04-06:
I'm closing this for lack of information. Feel free to reopen it if you're willing to provide more details.

closed 2023-04-06 by capntrips