capntrips/KernelFlasher #15: Error `Invalid boot.img, no ramdisk or kernel found` on Pixel 8

li-yifei, 2023-10-12:
IMO the boot partition or boot image itself changed a lot on devices launched with android 14

## Env
app version: 1.0.0-alpha16
root: Magisk 26301
phone: Pixel 8 Pro with 2023 1005 security update
stock kernel verison:  

> 5.15.110-android14-11-ga6d7915820a0-ab10726252
> #1 Mon Aug 28 18:42:09 UTC 2023

### Screenshot

[image]

### Boot Image sample

[p8p_stock_boot.zip](https://github.com/capntrips/KernelFlasher/files/12877179/p8p_stock_boot.zip)

capntrips, 2023-10-12:
`magiskboot` unpacks the `kernel` file from the image your zip without issue, so that doesn't appear to be the issue. It is also able to unpack `ramdisk.cpio` from the `init_boot` image in the in factory image.

Is there anything interesting in `logcat` when you get the error?
Does `/dev/block/by-name/boot_a` exist on the device? What about `/dev/block/by-name/init_boot_a`?
Does `httools dump boot` return an `fstab` entry? What about `httools dump init_boot`?

If everything above is normal, dumps from the the `boot` and `init_boot` partitions on both slots would be helpful.

capntrips, 2023-10-12:
> ```
> not executable: 32-bit ELF file
> ```

That's likely the problem. Here is a build with 64-bit versions of `httools_static`, `lptools_static`, and `magiskboot` included:

[KernelFlasher-v1.0.0-alpha16+arm64.apk.gz](https://github.com/capntrips/KernelFlasher/files/12880413/KernelFlasher-v1.0.0-alpha16%2Barm64.apk.gz)

On a related note, it appears that no version of `magiskboot` after v25.2 has a working `cpio sha1` command, so I'll see about opening an issue or PR about that in the next few days.

capntrips, 2023-10-12:
closing with the release of `v1.0.0-alpha17`

closed 2023-10-12 by capntrips

kvn1351, 2023-10-12:
Same issue on a Pixel 7 running A14 and alpha17. Hasn't ever been a problem before.

capntrips, 2023-10-13:
> it's specifically an issue with Raviantah's Pixel 7 kernel

Can you share a dump of the problematic `init_boot` partition?

Anything interesting in the logcat?

> Either way, slot B (stock kernel) should be found no? It did before.

When you say before, do you mean one of the previous releases? or specifically `v1.0.0-alpha16+arm64` that I shared above? I was toying with the ability to work with bad images in that build, but I'll have to make it a bit more robust before I can make a release with it.

a447948814, 2023-11-17:
小米14 安卓14也有这个问题

capntrips, 2023-11-17:
> Xiaomi 14 and Android 14 also have this problem

Can you share a dump of the problematic `init_boot` partition?

Anything interesting in the logcat?

a447948814, 2023-11-17:
> > Xiaomi 14 and Android 14 also have this problem
> 
> Can you share a dump of the problematic `init_boot` partition?
> 
> Anything interesting in the logcat?

我可以配合你解决此问题，但是我不太明白怎么给你dump文件，我提交了一个新的issue。（您的应用在中国玩家圈子里是非常受欢迎的，感谢您）