capntrips/KernelFlasher #16: 小米14安卓14打开alpha18提示Invalid boot.img, no ramdisk orkernel found

a447948814, 2023-11-17:
小米14手机，安卓14，alpha18打开就提示：
Invalid boot.img, no ramdisk orkernel found
[image]

a447948814, 2023-11-17:
使用网友二次修改的版本可以正常工作，我可以提供给你检查一下：https://taichi.lanzoum.com/inY1C0sxe09a
我希望最新版本可以修复这个问题

capntrips, 2023-11-17:
> The second revised version of the netizen can work normally, and I can provide it for you to check

I'm guessing something was lost in translation there. As noted in #15, dumps of the relevant partitions would be helpful.

Also the link didn't work for me, though at a glance it just appears to be the app's APK.

> I can cooperate with you to solve this problem, but I don't quite understand how to give you the dump file, I submitted a new issue. （Your application is very popular in the circle of Chinese players, thank you ）

From [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89113727), if you want some additional context:

```bash
adb shell
su
cd /data/local/tmp
dd if=/dev/block/by-name/boot_a of=boot_a.img
dd if=/dev/block/by-name/boot_b of=boot_b.img
dd if=/dev/block/by-name/init_boot_a of=init_boot_a.img
dd if=/dev/block/by-name/init_boot_b of=init_boot_b.img
exit
exit
adb pull /data/local/tmp/boot_a.img
adb pull /data/local/tmp/boot_b.img
adb pull /data/local/tmp/init_boot_a.img
adb pull /data/local/tmp/init_boot_b.img
```

a447948814, 2023-11-17:
1、那个apk是中国的技术爱好者基于您的官方版本alpha13作出的修改。
2、看了你的adb代码，您是希望我把boot.img和init_boot.img提供给你吗？是官方系统的原版boot还是我修补后刷入手机的boot？
3、我再次尝试用小米11安卓13安装了您的alpha18软件（他的root使用的是magis26.4+修补boot.img的方式），显示如下（似乎没有识别出已装载的卡槽）
[image]

capntrips, 2023-11-17:
> After reading your adb code, do you want me to provide you with boot.img and init_boot.img？

If you have 4 files, I really only need the `init_boot` ones, but I need both of them (`a` and `b`).

> Is it the original boot of the official system or the boot I brushed into my phone after repairing？

I have no way of knowing what you currently have installed on your device. You'll have to tell me.

I assume you made the dumps on the Xiaomi 14 Android 14 and you're still experiencing the issue with Kernel Flasher?

> I tried again to install your alpha18 software （ with Xiaomi 11 Android 13. His root used magis26.4+ to repair boot.img ）, showing the following （ does not seem to recognize the loaded Card slot ）

I'm not seeing any obvious issues in your screenshot. Can you elaborate on the issue you're experiencing?

P.S. It would be helpful to me if you would go ahead and translate your responses to English. As-is, if I don't directly quote every part of your message, I'll have to translate it any time I need to go back to review something.

a447948814, 2023-11-17:
I will provide you with three files, please keep an eye on their file names.

[xiaomi14pro-hyperOS23.11.13.zip](https://github.com/capntrips/KernelFlasher/files/13390439/xiaomi14pro-hyperOS23.11.13.zip)

a447948814, 2023-11-17:
There are currently two issues:
1. Xiaomi 14 Android 14 reported an error using the latest alpha18 version: Invalid boot.img, no ramdisk or kernel found (I have provided you with three files).
2. Xiaomi 11 Android 13 uses the latest Alpha18 version, although the main interface displays normally, the data of the A card slot is not recognized and does not show that it has been hung.

capntrips, 2023-11-17:
I'm able to run Kernel Flasher with the official and repair images you provided. To confirm, both of those images are currently flashed on the device, and one is  `init_boot_a` and the other is `init_boot_b`?

a447948814, 2023-11-17:
My current problem is that when I open the software, it displays an error and I am unable to proceed with subsequent operations. I am unable to scan the AK3 package.The software is unable to display the normal functional main interface.

capntrips, 2023-11-17:
The error is due to being unable to read or process one or both of `init_boot_a` or `init_boot_b`. [Here](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1815738693) and [here](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1815823048) I asked for dumps of those partitions _from your device_.

I ran a test with the two `init_boot` files you provided me, and Kernel Flasher was able to process both the files you provided me. [Here](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1816147958) I asked you to verify that the two files you provided me are the current `init_boot_a` or `init_boot_b` on your device. Is this the case? If not, can you provide me with those two files?

a447948814, 2023-11-20:
> The error is due to being unable to read or process one or both of `init_boot_a` or `init_boot_b`. [Here](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1815738693) and [here](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1815823048) I asked for dumps of those partitions _from your device_.
> 
> I ran a test with the two `init_boot` files you provided me, and Kernel Flasher was able to process both the files you provided me. [Here](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1816147958) I asked you to verify that the two files you provided me are the current `init_boot_a` or `init_boot_b` on your device. Is this the case? If not, can you provide me with those two dumps?

The file I provided you is the one I am currently using on my phone. I cannot provide both - A and - B files because there is only one file, either in slot A or slot B. In short, alpha18 cannot read the initboot file in slot A or slot B that I am using. I don't know why.

a447948814, 2023-11-20:
I conducted tests and found that：

alpha18-alpha15 ❌ Both prompt for errors
alpha14-alpha13 ✅ Can be used normally

Earlier versions did not test them。

a447948814, 2023-11-20:
I will now provide you with the file: boot_ a.img (and attach a screenshot of my interface using alpha14)

[image]

[boot_a.zip](https://github.com/capntrips/KernelFlasher/files/13417266/boot_a.zip)

capntrips, 2023-11-20:
> I cannot provide both - A and - B files because there is only one file, either in slot A or slot B.

You can retrieve `init_boot_a` and `init_boot_b` using `adb`, as I noted previously:

```bash
adb shell
su
dd if=/dev/block/by-name/init_boot_a of=/data/local/tmp/init_boot_a.img
dd if=/dev/block/by-name/init_boot_b of=/data/local/tmp/init_boot_b.img
exit
exit
adb pull /data/local/tmp/init_boot_a.img
adb pull /data/local/tmp/init_boot_b.img
```

I cannot debug this issue without those dumps. 

> I conducted tests and found that：
> 
> alpha18-alpha15 ❌ Both prompt for errors
> alpha14-alpha13 ✅ Can be used normally

There was a bug in alpha13 and alpha14 that erroneously checked the active slot rather than the inactive one.

a447948814, 2023-11-21:
> I cannot debug this issue without those dumps.
--Once again, would you like `boot_a.img` and `boot_b.img`, or `init_boot_a.img` and `init_boot_b.img`?

> There was a bug in alpha13 and alpha14 that erroneously checked the active slot rather than the inactive one.
--However, alpha13 and alpha14 do indeed work properly on my device.

capntrips, 2023-11-21:
`init_boot_a` and `init_boot_b`

a447948814, 2023-11-21:
> `init_boot_a` and `init_boot_b`

ok，I am doing，but，I am currently using “kernel su”, which seems to work through `boot. img`. I thought I wanted to provide you with `boot. img`，Please wait a moment

a447948814, 2023-11-21:
What should I do? It seems that adb is unable to obtain su permission，I am using the root mode of kernel su
[image]

capntrips, 2023-11-21:
> What should I do? It seems that adb is unable to obtain su permission，I am using the root mode of kernel su
> ```
> shennong:/ $ su
> /system/bin/sh: su: inaccessible or not found
> ```

I don't know anything about KernelSU. From [this](https://github.com/tiann/KernelSU/issues/972) it seems that it should normally be possible.

a447948814, 2023-11-21:
> I don't know anything about KernelSU. From [this](https://github.com/tiann/KernelSU/issues/972) it seems that it should normally be possible.

https://kernelsu.org/
Can I directly extract init_boot.img from the system ROM package?

capntrips, 2023-11-21:
> Can I directly extract init_boot.img from the ROM package?

No. The point is that issue is specific to your device. I assume the `init_boot` images from the zip you previously provided cover any issues with the image from the ROM package anyway.

I'll repeat my note from [above](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1816147958), just in case that can resolve your issue without further debugging:

> What I suspect is the device is new and the second slot is still completely blank (see [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89115144) for an example). If that is the case, a simple workaround would be to sideload an OTA to populate the second slot. Alternatively, you can use the test build linked in that post, which will allow Kernel Flasher to run with a bad slot.

Alternatively, you may be able to dump the `init_boot` images from each slot with alpha13 or alpha14, but I'm not positive that it will dump the correct image for the inactive slot.

tiann, 2023-11-21:
This commit may be helpful: https://github.com/capntrips/KernelFlasher/pull/11/commits/d1623fe0ba8da17609aa859171ce1a36655625fe

a447948814, 2023-11-21:
> > Can I directly extract init_boot.img from the ROM package?
> 
> No. The point is that issue is specific to your device. I assume the `init_boot` images from the zip you previously provided would have uncovered any issues with the image from the ROM package anyway.
> 
> I'll repeat my note from [above](https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1816147958), just in case that can resolve your issue without further debugging:
> 
> > What I suspect is the device is new and the second slot is still completely blank (see [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89115144) for an example). If that is the case, a simple workaround would be to sideload an OTA to populate the second slot. Alternatively, you can use the test build linked in that post, which will allow Kernel Flasher to run with a bad slot.
> 
> Alternatively, you may be able to dump the `init_boot` images from each slot with alpha13 or alpha14, but I'm not positive that it will dump the correct image for the inactive slot.

(I have resolved the permission issue by manually granting the shell program root permission on my phone.)
The following are the init_boot_a and init_boot_b files that I exported using the adb command according to your requirements. 
[adb to init_boot_A and init_boot_B.zip](https://github.com/capntrips/KernelFlasher/files/13421803/adb.to.init_boot_A.and.init_boot_B.zip)
(By the way, you can take a look at what Tinann said, which may be helpful to you.)

capntrips, 2023-11-21:
As was the case [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89115144) `init_boot_b` is empty. This must be a new device. If you flash a build of your OS to slot B, it should resolve your issue. I'm going to close this issue, but if it persists after you install the OS to slot B, feel free to reopen.

P.S. I'm almost certainly going to have to rework the initialization code of the app at some point, but it won't be in the next few days.

closed 2023-11-21 by capntrips

a447948814, 2023-11-21:
thankyou，

> As was the case [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89115144) `init_boot_b` is empty. This must be a new device. If you flash a build of your OS to slot B, it should [...]

I have updated the system to another slot B, and before restarting my phone, I need to also flash the AK3 zip file into another slot B. However, according to my recollection, this issue seems to still exist because at this point, slot A becomes empty for your application. It seems that it can only recognize the current card slot in the system and will never be able to recognize another card slot.（I'm not very sure, let me observe again）This also needs to be helpful to you：https://github.com/capntrips/KernelFlasher/issues/16#issuecomment-1820156371

capntrips, 2023-11-21:
Flashing to the inactive slot was removed in alpha03. I don't immediately recall what prompted that change, but if it is to be readded at some point in the future, it will be with AK3 functions, rather than `resetprop`. Both of the changes in that comment were discussed in #11, if you want to review the logic behind why they weren't implemented as-is. If you need that capability in the immediate future, I imagine it is implemented in the build in that fork.