capntrips/KernelFlasher #20: Xiaomi Mi 10T A only device doesn't work

matthew777777, 2023-12-17:
Hi,
I tried to use the app but it doesn't work with A only devices like Apollo 
[image]

capntrips, 2023-12-17:
What are the contents of your `/dev/block/by-name`?

matthew777777, 2023-12-17:
> What are the contents of your `/dev/block/by-name`?

Here they are 
[image]

matthew777777, 2023-12-17:
> > What are the contents of your `/dev/block/by-name`?
> 
> Here they are ![Screenshot_20231217-202431](https://private-user-images.githubusercontent.com/134064721/291098826-376e8abc-0131-42be-8cd4-b7d349ad66ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI4NDgxMDksIm5iZiI6MTcwMjg0NzgwOSwicGF0aCI6Ii8xMzQwNjQ3MjEvMjkxMDk4ODI2LTM3NmU4YWJjLTAxMzEtNDJiZS04Y2Q0LWI3ZDM0OWFkNjZhYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxN1QyMTE2NDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZjAxZGY2ZGQ5MDgyNmNiYTUyYWI2NWQ0NzkyYTU2NDBlMTkxZDVjNjE2ZjI0MWRjMzNkZTg1YzQ3NzhkYmE1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Mwf5XVhlOEt_gQLCIzuzCz7M895-x3zLd9bsC4OD7n8)

Latest build doesn't work too here is logcat may help
[image]

[17_12-22-19-13_524.log](https://github.com/capntrips/KernelFlasher/files/13697800/17_12-22-19-13_524.log)

capntrips, 2023-12-17:
I took a quick peek at what would be required, and it isn't something I'm currently willing to take on without access to a non-A/B device.

matthew777777, 2023-12-17:
> I took a quick peek at what would be required, and it isn't something I'm currently willing to take on without access to a non-A/B device.

Thank you for your answer.
Shalom Shalom

capntrips, 2024-01-22:
A test build, `v1.0.0-alpha019+allow-non-ab`, is now available on the releases page. See [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89290959) for additional details.

closed 2024-01-22 by capntrips

matthew777777, 2024-01-22:
> A test build, `v1.0.0-alpha019+allow-non-ab`, is now available on the releases page. See [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89290959) for additional details.

Wow 😊 Thank you it works. I see you are wheel of time fan as me 😊
[image]

capntrips, 2024-01-22:
That's great, and yes, I'm a big fan of the books. I've used the cover of the first one on my profile for like seven years, and I think you're the first to mention it.