capntrips/KernelFlasher #21: Crashing on Android 14

ARandomPerson7, 2023-12-28:
- Device: Samsung Galaxy s23 Plus
- ROM: Stock
- Root: Magisk
- KernelFlasher version: v1.0.0-alpha19

App crashing after granting root permission. Log is below:

`
Process: com.github.capntrips.kernelflasher, PID: 12248
java.lang.RuntimeException: Error receiving broadcast Intent { 
act=com.topjohnwu.superuser.RECEIVER_BROADCAST flg=0x400010 pkg=com.github.capntrips.kernelflasher (has extras) 
} in com.topjohnwu.superuser.internal.RootServiceManager$ServiceReceiver@ea5ad8e
at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0(LoadedApk.java:1884)
at android.app.LoadedApk$ReceiverDispatcher$Args.$r8$lambda$gDuJqgxY6Zb-ifyeubKeivTLAwk(Unknown Source:0)
at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
at android.os.Handler.handleCallback(Handler.java:958)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loopOnce(Looper.java:230)
at android.os.Looper.loop(Looper.java:319)
at android.app.ActivityThread.main(ActivityThread.java:8893)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:608)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
`

capntrips, 2024-01-09:
Is there more in the log? If I'm not mistaken, this is the result of the service failing to start, but I'm not seeing the backtrace from that failure.

ARandomPerson7, 2024-01-12:
My bad, here is the full backtrace. Sorry if the formatting isn't good, not sure how to make it look good here.

`com.github.capntrips.kernelflasher   E  FATAL EXCEPTION: main
                                                                                                    Process: com.github.capntrips.kernelflasher, PID: 6815
                                                                                                    java.lang.RuntimeException: Error receiving broadcast Intent { act=com.topjohnwu.superuser.RECEIVER_BROADCAST flg=0x400010 pkg=com.github.capntrips.kernelflasher (has extras) } in com.topjohnwu.superuser.internal.RootServiceManager$ServiceReceiver@5c49631
                                                                                                    	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0(LoadedApk.java:1884)
                                                                                                    	at android.app.LoadedApk$ReceiverDispatcher$Args.$r8$lambda$gDuJqgxY6Zb-ifyeubKeivTLAwk(Unknown Source:0)
                                                                                                    	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:958)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:230)
                                                                                                    	at android.os.Looper.loop(Looper.java:319)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8893)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:608)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
                                                                                                    Caused by: java.lang.NullPointerException
                                                                                                    	at com.github.capntrips.kernelflasher.ui.screens.main.MainViewModel.<init>(MainViewModel.kt:60)
                                                                                                    	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2$1.invoke(MainActivity.kt:185)
                                                                                                    	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2$1.invoke(MainActivity.kt:183)
                                                                                                    	at androidx.lifecycle.viewmodel.InitializerViewModelFactory.create(InitializerViewModelFactory.kt:107)
                                                                                                    	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:187)
                                                                                                    	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:153)
                                                                                                    	at androidx.lifecycle.viewmodel.compose.ViewModelKt.get(ViewModel.kt:215)
                                                                                                    	at androidx.lifecycle.viewmodel.compose.ViewModelKt.viewModel(ViewModel.kt:156)
                                                                                                    	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2.invoke(MainActivity.kt:332)
                                                                                                    	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2.invoke(MainActivity.kt:181)
                                                                                                    	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:108)
                                                                                                    	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:35)
                                                                                                    	at androidx.compose.ui.platform.ComposeView.Content(ComposeView.android.kt:428)
                                                                                                    	at androidx.compose.ui.platform.AbstractComposeView$ensureCompositionCreated$1.invoke(ComposeView.android.kt:252)
[... cut ...]
                                                                                                    	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1$2.invoke(Wrapper.android.kt:158)
                                                                                                    	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1$2.invoke(Wrapper.android.kt:157)
                                                                                                    	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:108)`

capntrips, 2024-01-13:
What are the contents of your `/dev/block/by-name`?

ARandomPerson7, 2024-01-14:
[image]

capntrips, 2024-01-14:
This appears to be a non-A/B device, and I don't currently plan to add support for them. You can follow #20 in case that changes.

capntrips, 2024-01-22:
A test build adding support for non-A/B devices is up. See [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89290959) for additional details.

closed 2024-01-22 by capntrips