capntrips/KernelFlasher #22: Crashes on OnePlus 3T

cherinyy, 2024-01-16:
Does Kernel Flasher not support Legacy Android devices?

- [logcat.zip](https://github.com/capntrips/KernelFlasher/files/13945263/logcat.zip)

---

Device Info: https://wiki.lineageos.org/devices/oneplus3/variant2/

capntrips, 2024-01-16:
Based on the exception, this appears to be a non-A/B device, and I don't currently plan to add support for them. You can follow #20 in case that changes.

Feel free to reopen if this is actually an A/B device. If so, you should be able to find `boot_a` and `boot_b` in `/dev/block/by-name`.

Since this has come up several times lately, I should probably update the app to handle it a bit more gracefully.

capntrips, 2024-01-22:
A test build adding support for non-A/B devices is up. See [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89290959) for additional details.

closed 2024-01-22 by capntrips