capntrips/KernelFlasher #23: The kernel cannot be found, what should I do?

Chuhe2399, 2024-01-19:
[image]

Chuhe2399, 2024-01-19:
[image]

capntrips, 2024-01-20:
First, what is the device? Is it an A/B device. If so, you should be able to find `boot_a` and `boot_b` in `/dev/block/by-name`. If not, it is unsupported. Is it brand new? If so, the issue might be that one slot is completely empty, in which case the solution is to install an OTA to populate the opposite slot. Otherwise, I'll need dumps of the relevant partitions. From [here](https://xdaforums.com/t/dev-kernel-flasher.4439887/post-89113727), if you want some additional context:

```bash
adb shell
su
cd /data/local/tmp
dd if=/dev/block/by-name/boot_a of=boot_a.img
dd if=/dev/block/by-name/boot_b of=boot_b.img
dd if=/dev/block/by-name/init_boot_a of=init_boot_a.img
dd if=/dev/block/by-name/init_boot_b of=init_boot_b.img
exit
exit
adb pull /data/local/tmp/boot_a.img
adb pull /data/local/tmp/boot_b.img
adb pull /data/local/tmp/init_boot_a.img
adb pull /data/local/tmp/init_boot_b.img
```

ltkun, 2024-01-20:
This happened when I upgrade a miui14 to Xiaomi hyperos.
Because Xiaomi will wipe up the slot which equipped the MIUI14.
My kernel flasher version is latest. And phone is Xiaomi 13 ultra.

capntrips, 2024-01-21:
A test build, v1.0.0-alpha19+allow-invalid, is now available on the releases page. If this does not resolve your issue, feel free to reopen.

closed 2024-01-21 by capntrips