capntrips/KernelFlasher #24: Crash after granting root permision

emmandesu, 2024-01-28:
java.lang.RuntimeException: Error receiving broadcast Intent { act=com.topjohnwu.superuser.RECEIVER_BROADCAST flg=0x400010 pkg=com.github.capntrips.kernelflasher (has extras) } in com.topjohnwu.superuser.internal.RootServiceManager$ServiceReceiver@56fa471
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(Unknown Source:382)
	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
	at android.os.Handler.handleCallback(Unknown Source:2)
	at android.os.Handler.dispatchMessage(Unknown Source:4)
	at android.os.Looper.loopOnce(Unknown Source:182)
	at android.os.Looper.loop(Unknown Source:82)
	at android.app.ActivityThread.main(Unknown Source:123)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(Unknown Source:11)
	at com.android.internal.os.ZygoteInit.main(Unknown Source:312)
Caused by: java.lang.NullPointerException
	at com.github.capntrips.kernelflasher.ui.screens.main.MainViewModel.<init>(MainViewModel.kt:60)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2$1.invoke(MainActivity.kt:185)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2$1.invoke(MainActivity.kt:183)
	at androidx.lifecycle.viewmodel.InitializerViewModelFactory.create(InitializerViewModelFactory.kt:107)
	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:187)
	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:153)
	at androidx.lifecycle.viewmodel.compose.ViewModelKt.get(ViewModel.kt:215)
	at androidx.lifecycle.viewmodel.compose.ViewModelKt.viewModel(ViewModel.kt:156)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2.invoke(MainActivity.kt:332)
	at com.github.capntrips.kernelflasher.MainActivity$onAidlConnected$2.invoke(MainActivity.kt:181)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:108)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:35)
	at androidx.compose.ui.platform.ComposeView.Content(ComposeView.android.kt:428)
	at androidx.compose.ui.platform.AbstractComposeView$ensureCompositionCreated$1.invoke(ComposeView.android.kt:252)
	at androidx.compose.ui.platform.AbstractComposeView$ensureCompositionCreated$1.invoke(ComposeView.android.kt:251)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:108)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:35)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.ui.platform.CompositionLocalsKt.ProvideCommonCompositionLocals(CompositionLocals.kt:195)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt$ProvideAndroidCompositionLocals$3.invoke(AndroidCompositionLocals.android.kt:119)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt$ProvideAndroidCompositionLocals$3.invoke(AndroidCompositionLocals.android.kt:118)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:108)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:35)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt.ProvideAndroidCompositionLocals(AndroidCompositionLocals.android.kt:110)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1$2.invoke(Wrapper.android.kt:158)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1$2.invoke(Wrapper.android.kt:157)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:108)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:35)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1.invoke(Wrapper.android.kt:157)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1$1.invoke(Wrapper.android.kt:142)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:108)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:35)
	at androidx.compose.runtime.ActualJvm_jvmKt.invokeComposable(ActualJvm.jvm.kt:78)
	at androidx.compose.runtime.ComposerImpl.doCompose(Composer.kt:3340)
	at androidx.compose.runtime.ComposerImpl.composeContent$runtime_release(Composer.kt:3273)
	at androidx.compose.runtime.CompositionImpl.composeContent(Composition.kt:588)
	at androidx.compose.runtime.Recomposer.composeInitial$runtime_release(Recomposer.kt:1013)
	at androidx.compose.runtime.CompositionImpl.setContent(Composition.kt:520)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:142)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:133)
	at androidx.compose.ui.platform.AndroidComposeView.setOnViewTreeOwnersAvailable(AndroidComposeView.android.kt:1191)
	at androidx.compose.ui.platform.WrappedComposition.setContent(Wrapper.android.kt:133)
	at androidx.compose.ui.platform.WrappedComposition.onStateChanged(Wrapper.android.kt:183)
	at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.kt:314)
	at androidx.lifecycle.LifecycleRegistry.addObserver(LifecycleRegistry.kt:192)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:140)
	at androidx.compose.ui.platform.WrappedComposition$setContent$1.invoke(Wrapper.android.kt:133)
	at androidx.compose.ui.platform.AndroidComposeView.onAttachedToWindow(AndroidComposeView.android.kt:1266)
	at android.view.View.dispatchAttachedToWindow(Unknown Source:82)
	at android.view.ViewGroup.dispatchAttachedToWindow(Unknown Source:7)
	at android.view.ViewGroup.dispatchAttachedToWindow(Unknown Source:36)
	at android.view.ViewGroup.addViewInner(Unknown Source:99)
	at android.view.ViewGroup.addView(Unknown Source:10)
	at android.view.ViewGroup.addView(Unknown Source:1)
	at com.android.internal.policy.PhoneWindow.setContentView(Unknown Source:43)
	at android.app.Activity.setContentView(Unknown Source:4)
	at androidx.activity.ComponentActivity.setContentView(ComponentActivity.java:456)
	at androidx.activity.compose.ComponentActivityKt.setContent(ComponentActivity.kt:70)
	at androidx.activity.compose.ComponentActivityKt.setContent$default(ComponentActivity.kt:51)
	at com.github.capntrips.kernelflasher.MainActivity.onAidlConnected(MainActivity.kt:181)
	at com.github.capntrips.kernelflasher.MainActivity$AidlConnection.onServiceConnected(MainActivity.kt:77)
	at com.topjohnwu.superuser.internal.RootServiceManager.lambda$bindInternal$2(RootServiceManager.java:267)
	at com.topjohnwu.superuser.internal.RootServiceManager$$ExternalSyntheticLambda2.run(Unknown Source:6)
	at com.topjohnwu.superuser.internal.UiThreadHandler.run(UiThreadHandler.java:31)
	at com.topjohnwu.superuser.internal.UiThreadHandler$$ExternalSyntheticLambda0.execute(Unknown Source:0)
	at com.topjohnwu.superuser.internal.RootServiceManager.bindInternal(RootServiceManager.java:267)
	at com.topjohnwu.superuser.internal.RootServiceManager.lambda$createBindTask$4$com-topjohnwu-superuser-internal-RootServiceManager(RootServiceManager.java:283)
	at com.topjohnwu.superuser.internal.RootServiceManager$$ExternalSyntheticLambda4.run(Unknown Source:8)
	at com.topjohnwu.superuser.internal.RootServiceManager$ServiceReceiver.onReceive(RootServiceManager.java:452)
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(Unknown Source:258)
	... 9 more

capntrips, 2024-01-28:
It looks like a non-A/B device from the exception. Have you tried the `v1.0.0-alpha019+allow-non-ab` test build?

capntrips, 2024-02-13:
I'm going to close this for lack of info, but feel free to reopen and provide additional details.

closed 2024-02-13 by capntrips