capntrips/KernelFlasher #27: Why should we check here

RimuruChan, 2024-03-17:
Apologies for any inconvenience. I'm curious about why we prefer to perform a check and use the SHA1 of a CPIO file instead of simply calculating the SHA1 of the partition itself?

https://github.com/capntrips/KernelFlasher/blob/1cbbba497f42755d6459a83d9a7d8ee14f550572/app/src/main/java/com/github/capntrips/kernelflasher/ui/screens/slot/SlotViewModel.kt#L115-L125

tstaylor7, 2024-03-17:
The original idea was to make it easy to identify the original boot image at a glance. The backups are also tied to that hash. A change to this functionality has been requested on XDA, as well. I plan to add the ability to toggle both via a setting menu, but I wouldn't expect it in the coming weeks.

capntrips, 2024-03-17:
The original idea was to keep up with the original boot image to avoid bootloops. The available backups are also tied to that value. A change to that functionality has been requested on XDA, as well. I plan to add toggles for both features on a settings screen, but I wouldn't expect it in the coming weeks.