capntrips/KernelFlasher #28: Invalid ramdisk in boot.img (Pixel 8 Pro, KernelSU)

scarlettekk, 2024-03-17:
Pixel 8 pro, android QPR Beta (AP21.240216.010)
When using the new kernel module install mode of KernelSU, the app does not load with the given error message

capntrips, 2024-03-17:
Can you share dumps of the `boot` and `init_boot` partitions?

```bash
adb shell
su
cd /data/local/tmp
dd if=/dev/block/by-name/boot_a of=boot_a.img
dd if=/dev/block/by-name/boot_b of=boot_b.img
dd if=/dev/block/by-name/init_boot_a of=init_boot_a.img
dd if=/dev/block/by-name/init_boot_b of=init_boot_b.img
exit
exit
adb pull /data/local/tmp/boot_a.img
adb pull /data/local/tmp/boot_b.img
adb pull /data/local/tmp/init_boot_a.img
adb pull /data/local/tmp/init_boot_b.img
```

Do you have any details on where the kernel module is installed? If it goes into a separate partition, can you share that dump, as well?

scarlettekk, 2024-03-17:
[a.zip](https://github.com/capntrips/KernelFlasher/files/14628600/a.zip)
[b.zip](https://github.com/capntrips/KernelFlasher/files/14628602/b.zip)
The module is installed by patching init_boot, other than that I am not sure how it works. The current slot is _a

capntrips, 2024-03-20:
A test build, `v1.0.0-alpha20+allow-errors`, is now available on the releases page. If this does not resolve your issue, feel free to reopen.

closed 2024-03-20 by capntrips

scarlettekk, 2024-03-20:
It loads now but still shows the warning on slot 1. Will it still work correctly?

capntrips, 2024-03-20:
Here are my notes from the XDA support thread:

> The `allow-errors` build will ignore any kernel or ramdisk errors (even on the active slot), which will hopefully allow it to work with KernelSU installed as a kernel module mode. This may have implications in other areas. For example, since it allows a null sha1, it may behave unexpectedly when attempting to make backups when there is a ramdisk error. I'll make an effort to add a settings screen with toggles for sha1 and error handling before I remove the pre-release tag.

Beyond that, I can only speculate, as I don't know any about how KernelSU works. The error message is only informational, and potential quirks with backups aside, there should be nothing that prevents Kernel Flasher from performing its operations. I have no idea if that ramdisk state has any implications for AK3.

Assuming no issues are reported, I may drop the error messages when I remove the pre-release tag.

xuangeyouneihan, 2024-07-03:
When I tried to restore the kernel with KernelSU LKM, it crashes, even though I installed the allow-error version

capntrips, 2024-07-03:
Can you share a logcat?

userosos, 2024-09-03:
Hi! How i can start KernelFlasher with KernelSU? If i set SU User in kernelSU manager i see in kernel flasher start window - invalid ramdisk in boot.img. I change default profile and it not helped - i can see differents errors (for differents profiles) in start - No root, or damaged boot.ini for example.