capntrips/KernelFlasher #33: Allow Apatch/Ksu to flash, instead of ramdisk only?

AlexS2412, 2024-06-22:
seems to be built for magisk only...

capntrips, 2024-07-03:
Can you share any additional info? While the app is capable of flashing partition images, as far as flashing goes, it mostly just sets up the environment to run AK3. Is AK3 able to perform the operations you need?

AlexS2412, 2024-07-03:
> Can you share any additional info? While the app is capable of flashing partition images, as far as flashing goes, it mostly just sets up the environment to run AK3. Is AK3 able to perform the operations you need?

the issue seems to be all ak3 zips look for magisk ramdisk, so when you are on apatch/ksu that lives in the kernel, the zip wants to find, and repack your ramdisk with the custom kernel image. obviously this is impossible with apatch. i tried to manually patch the custom boot.imgs for a couple kernels with apatch and still end up bricking anyway. it just seems like apatch is too immature for custom kernels or something. that oe lack of documentation.

YuukyuuKikan, 2024-07-15:
> the issue seems to be all ak3 zips look for magisk ramdisk, so when you are on apatch/ksu that lives in the kernel, the zip wants to find, and repack your ramdisk with the custom kernel image. [...]

Reading through the KSU issues, I agree with AlexS2412.

Seems AK3 itself has compability issues with KernelSU if "it mostly just sets up the environment to run AK3".

closed 2025-06-19 by capntrips