capntrips/KernelFlasher #36: 'Updates' feature enquiry

DestroyerBDT, 2025-03-02:
I would like to enquire about the *Updates* feature. Is it like [Obtainium](https://f-droid.org/packages/dev.imranr.obtainium.fdroid), which can get updates from a GitHub repository and use Regex to filter the appropriate kernel?

capntrips, 2025-03-06:
It may not function identically to FKM or SmartPack, but [this](https://smartpack.github.io/kerneldownloads/) is the basic idea. If a kernel developer provides the link (e.g. [this](https://raw.githubusercontent.com/engstk/kernelfkm/master/updater_pxl7a.json)), it can be used to check for updates to their kernel.

closed 2025-03-06 by capntrips

DestroyerBDT, 2025-03-07:
> It may not function identically to FKM or SmartPack, but [this](https://smartpack.github.io/kerneldownloads/) is the basic idea. If a kernel developer provides the link (e.g. [...]

Thank you, this was very helpful information