capntrips/KernelFlasher #40: There was a problem parsing the package.

psychoreactorq, 2025-05-09:
I tried KernelFlasher-v1.0.0-alpha20.apk and KernelFlasher-v1.0.0-alpha20 allow-errors.2.apk
The installer failed with the error message in title. I am using a Moto g6 running Android 9 that is not rooted.

Trying 'adb install' provided additional data:
> adb: failed to install KernelFlasher-v1.0.0-alpha20+allow-errors.2.apk: Failure [INSTALL_FAILED_OLDER_SDK: Failed parse during installPackageLI: /data/app/vmdl157701337.tmp/base.apk (at Binary XML file line #7): Requires newer sdk version #29 (current version is #28)]

Could you please list the minimum compatible Android version in the documentation?

I was hoping to find something that can extract partition images as low as Android 4, especially for devices without root.

Thanks!

capntrips, 2025-06-19:
Here is a test build for sdk28: 

[KernelFlasher-v1.0.0-alpha20+allow-errors.sdk28.tar.gz](https://github.com/user-attachments/files/20823934/KernelFlasher-v1.0.0-alpha20%2Ballow-errors.sdk28.tar.gz)

It was built against the `allow-errors` branch and is working fine for me on Android 16. Let me know if it's working on Android 9.

**Note:** Github won't allow me to directly upload an APK here, so I included it and the patch in a tarball.

yan12125, 2025-06-21:
> Let me know if it's working on Android 9.

Thank you! It seems working fine. I can flash a boot image.

> I wouldn't expect the compiled version to be identical to source. The one linked is identical to my local copy.

Got it, thanks for clarification.

closed 2025-08-19 by capntrips